guysoft/OctoPi #837: OctoPi 1.1.0 RC1 Status

guysoft, 2024-11-22:
First release candidate for OctoPi 1.1.0

There are both 32bit and 64bit images available for Raspberry Pi

There is also a new beta release for [Le Potato AML-S905X-CC](https://libre.computer/products/aml-s905x-cc/).
Its beta because I is not as tested as OctoPi for Raspberrypi. And I hope that having it available might let people play with it and tell me what can be done to get the wifi and cameras working correctly. 

This image brings support to Raspberry Pi 5. It also drops wpa-supplicant support due to https://github.com/raspberrypi/bookworm-feedback/issues/72
Please try the release candidate so we know it works.

32bit armf:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly/2024-11-22_2024-11-19-octopi-bookworm-armhf-lite-1.1.0.zip

Md5: [``173a6b93942e83c960d84d70f6c6f778``](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2024-11-11_2024-10-22-octopi-bookworm-armhf-lite-1.1.0.zip.md5).

64bit arm64/aarch64:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2024-11-22_2024-11-19-octopi-bookworm-arm64-lite-1.1.0.zip

Md5: [``5344de5850bc402da5e741e96dcf3eef``](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2024-11-22_2024-11-19-octopi-bookworm-arm64-lite-1.1.0.zip.md5).

Let Potato AML-S905X-CC (based on debian):
https://unofficialpi.org/Distros/OctoPi/debian_lepotato/nightly/2024-11-22_debian-12-base-arm64+aml-s905x-cc-1.1.0.zip
Md5: [``5c892cb5b187ad79612f39597c214054``](https://unofficialpi.org/Distros/OctoPi/debian_lepotato/nightly/2024-11-22_debian-12-base-arm64+aml-s905x-cc-1.1.0.zip.md5)

### Changes in the image

* Raspberry Pi 5 support https://github.com/guysoft/OctoPi/issues/823
* Beta support for Le Potato AML-S905X-CC
* Images uses Debian bookworm
* Due to bookworkm ``/boot/config.txt`` has moved to ``/boot/firmware/config.txt`` See https://github.com/guysoft/CustomPiOS/issues/220
* Support changing default user name on rpi-imager
    * Move octoprint python virtualenv to /opt/custompios/oprint and add a symlink https://github.com/guysoft/OctoPi/commit/e62bada2c4e91372ace724f7839ab61e34bfb824
* Dropped support for wpa supplicant due to https://github.com/raspberrypi/bookworm-feedback/issues/72
Please try the release candidate so we know it works.
* New NetworkManager headless support added, its pretty basic for now since it is expected most people configure the wifi via rpi-imager. There is a new ``wifi.nmconnection`` file in the ``/boot/firmware`` folder that lets you configure wifi headless. Its more limited at the moment than the wpa supplicant version
* Remove camera ID 046d:0825 Fixes https://github.com/guysoft/OctoPi/issues/759

foosel, 2024-11-25:
Did some initial testing and here's what I found:

- Image size is now 3.7GB. That's a *huge* increase compared to the 2.9GB of 1.0.0 and 2.5GB of 0.18. Is there any reason for that? Has the lite image grown that much? Is there any way this can be shrinked back again a bit?
	- 64bit image is even bigger at 4.1 GB
- Tested (against the testrig, using the RPi imager approach for customization of wifi & host):
	- 32bit on Pi3 w/ USB webcam: booted, setup wizard completed, camera works
	- 64bit on Pi3 w/ USB webcam: booted, setup wizard completed, camera works
	- 32bit on Pi3 w/ RPi cam (not v3): booted, setup wizard completed, **camera DOESN'T work**, `vcgencmd get_camera` says it's not detected
	- 64bit on Pi3 w/ RPi cam (not v3): booted, setup wizard completed, **camera DOESN'T work**, `vcgencmd get_camera` says it's not detected
	- 64bit on Pi5 w/o cam: booted, setup wizard completed, no camera to test
- Just to make sure that the issue with the RPi camera wasn't caused by my hardware suddenly dying I also tested against an OctoPi-UpToDate build (and thus OctoPi 1.0.0), and here it still works just fine, and `vcgencmd get_camera` also reports the camera as detected. 

So we do have an issue here, as `webcamd` no longer seems to work now even with the old RPi Cams that still worked fine under bullseye/1.0.0. I'd consider that a blocker. Alas, the new camera stack is currently also not an option yet, as there's a quite problematic bug in the underlying webcam server there as well, and I've not been able to get feedback from the maintainer on it and am currently starting to lean towards switching the server once more due to this issue 😕

guysoft, 2024-11-25:
I found that the wifi.nmconnection seems to have radio disabled at boot for 5ghz and needs to fix that somehow

guysoft, 2024-12-01:
If anyone has any idea why the cameras before V3 don't work it would help me fix it and have RC2 out

00ihsan, 2024-12-02:
I still do not see octoprint on the rpi imager with pi 5 selected.

guysoft, 2024-12-02:
@00ihsan Because its not released yet, its a candidate.
You can install it using rpi imager with this command:
```
rpi-imager --repo https://unofficialpi.org/rpi-imager/rpi-imager-octopi.json
```

00ihsan, 2024-12-02:
@guysoft thanks for the quick reply!

any idea when its released/ETA? No pressure!

guysoft, 2024-12-02:
@00ihsan I am currently block on cameras not detected. I didn't get to plugging one to a pi to debug. I hope in a few weeks. But any help on that would speed it up. Also depends if we need an RC3.

00ihsan, 2024-12-03:
Copy that! I’ll see if i can help with that.

Met vriendelijke groet,
Ihsan Topcu

guysoft, 2024-12-11:
Ok, looks like if I run ``vcgencmd get_camera`` I get:
```
supported=1 detected=0, libcamera interfaces=1
```
The camera can take pictures with:
```
rpicam-still -o /tmp/first_image.jpg
```

If I switch in ``/boot/firmware/config.txt`` 
from ``camera_auto_detect=1`` to ``camera_auto_detect=0``

I get when running ``vcgencmd get_camera``:
```
supported=1 detected=1, libcamera interfaces=0
```

Then the camera works normally. So will change that.

guysoft, 2024-12-16:
Looks like this commit fixes the network manager issue:
https://github.com/guysoft/CustomPiOS/commit/e553cad7e09410bf27a2083211d6e606d76efe21

So this means all reported RC1 issues have been fixed and can build RC2

The only thing is the size of the image, will try see what grew

guysoft, 2024-12-16:
Took a peek with ncdu on both images. (``mount2`` is the old image, ``mount`` is the new)
[image]

[image]
Does not look something grew a lot in particular. the python env moved to ``/opt``, and it looks like overall the binaries grew. But nothing I can put my finger on.

Here is usr:
[image]
[image]

Taomyn, 2024-12-17:
Was hoping to try this some time this week, so I tried to use the RPi imager to grab the latest, but it's a bit confusing as to which 64bit one is which as some are duplicated and :

[image]

I'm using the Windows imager with the command: `"C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe" --repo https://octopi.gnethomelinux.com/rpi-imager/rpi-imager-octopi.json` the same happens with the mirror link.

guysoft, 2024-12-30:
New RC2 is practically ready. I am waiting for end of holidays to publish it

guysoft, 2025-01-07:
Use RC2
https://github.com/guysoft/OctoPi/issues/842

closed 2025-01-07 by guysoft