guysoft/OctoPi #839: Do you remove the NTP functionality from the base OS, or has the NTP package/command changed?

allanonmage, 2024-12-23:
I'm tracking down an issue with the "remember me" cookie, and two new instances of Octoprint that I've set up via OctoPi have dates and times that are way out, and do not have ntpstat, so presumably they are not set up with time sync.  I know the included packages, their names, and all sorts of things change over time when it comes to the Raspberry Pi, this wouldn't be the first time a different package was selected for the OS, but even a forum post from 2020 references this specific package.  I would presume that you wouldn't go out of your way to remove NTP for your image, but regardless, mine are not keeping time well and I'm not sure why.

I've never had to worry about time sync in ~14 years of using OctoPi, so I suspect there's something with the version I downloaded (OctoPrint 1.10.3 Python 3.9.2 OctoPi* 1.0.0 (build 2024.11.05.092907))

guysoft, 2025-01-02:
There was this change but it was not merged yet:
https://github.com/guysoft/CustomPiOS/pull/241

The thing is that OctoPi 1.0.0 was released in 2022 - you are rebuilding. it somehow? What is the build number you provided?

If you are rebuilding there was a move to CustomPiOS v2 which might have done something, you can try the CustomPiOS ``release/1`` branch and see if that makes a change.

allanonmage, 2025-01-02:
Turns out it was an issue with DNS on my network, so the time sync couldn't make a connection, thereby causing the "remember Me" checkbox to fail.

I'm not sure what NTP package is included, but I found another random NTP related command and could verify it was working.

idk what you're talking about rebuilding; I flashed the image, I didn't build from source.

guysoft, 2025-01-05:
Ok, i thought you had some other issue with a cusom built one, indeed NTP works. closing.

closed 2025-01-05 by guysoft