guysoft/OctoPi #842: OctoPi 1.1.0 RC2 Status

guysoft, 2025-01-07:
Second release candidate for OctoPi 1.1.0

There are both 32bit and 64bit images available for Raspberry Pi

There is also a new beta release for [Le Potato AML-S905X-CC](https://libre.computer/products/aml-s905x-cc/).
Its beta because I is not as tested as OctoPi for Raspberrypi. And I hope that having it available might let people play with it and tell me what can be done to get the wifi and cameras working correctly. 

This image brings support to Raspberry Pi 5. It also drops wpa-supplicant support due to https://github.com/raspberrypi/bookworm-feedback/issues/72
Please try the release candidate so we know it works.

32bit armf:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly/2025-01-06_2024-11-19-octopi-bookworm-armhf-lite-1.1.0.zip

Md5: [``64c2d6c5d0eb82d25007511395866901``](https://unofficialpi.org/Distros/OctoPi/nightly/2025-01-06_2024-11-19-octopi-bookworm-armhf-lite-1.1.0.zip.md5).

64bit arm64/aarch64:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2025-01-06_2024-11-19-octopi-bookworm-arm64-lite-1.1.0.zip

Md5: [``414fd01b4727bd387a4eb92e306bcf53``](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2025-01-06_2024-11-19-octopi-bookworm-arm64-lite-1.1.0.zip.md5).

Let Potato AML-S905X-CC (based on debian):
https://unofficialpi.org/Distros/OctoPi/debian_lepotato/nightly/2025-01-06_debian-12-base-arm64+aml-s905x-cc-1.1.0.zip
Md5: [``3f2841fb39140e160f6cd4fce89137e3``](https://unofficialpi.org/Distros/OctoPi/debian_lepotato/nightly/2025-01-06_debian-12-base-arm64+aml-s905x-cc-1.1.0.zip.md5)

### Changes in the image

* Raspberry Pi 5 support https://github.com/guysoft/OctoPi/issues/823
* Beta support for Le Potato AML-S905X-CC
* Images uses Debian bookworm
* Due to bookworkm ``/boot/config.txt`` has moved to ``/boot/firmware/config.txt`` See https://github.com/guysoft/CustomPiOS/issues/220
* Support changing default user name on rpi-imager
    * Move octoprint python virtualenv to /opt/custompios/oprint and add a symlink https://github.com/guysoft/OctoPi/commit/e62bada2c4e91372ace724f7839ab61e34bfb824
* Dropped support for wpa supplicant due to https://github.com/raspberrypi/bookworm-feedback/issues/72
Please try the release candidate so we know it works.
* New NetworkManager headless support added, its pretty basic for now since it is expected most people configure the wifi via rpi-imager. There is a new ``wifi.nmconnection`` file in the ``/boot/firmware`` folder that lets you configure wifi headless. Its more limited at the moment than the wpa supplicant version
* Remove camera ID 046d:0825 Fixes https://github.com/guysoft/OctoPi/issues/759

Changes from RC1:
* Fixed issue with webcam not working https://github.com/guysoft/OctoPi/issues/837#issuecomment-2536532941
* Fixed wifi blocking https://github.com/guysoft/CustomPiOS/commit/e553cad7e09410bf27a2083211d6e606d76efe21
* Network check has been fixed #840 (Thanks @david-forster10)

bengalih, 2025-01-21:
How is the network check working in the RC?
Maybe I'm not looking at the right code base, but it appears `enable_network_monitor` is still looking for wpa_supplicant configuration?
I would like to know how the RC attempts to recover via Network Manager when connectivity is lost.

guysoft, 2025-01-23:
@bengalih yes you are right, this needs to be updated:
https://github.com/guysoft/OctoPi/blob/252e4f10b31a4251687213b1b4770a5e2c5845aa/src/modules/octopi/filesystem/root/usr/local/bin/networkcheck#L10

guysoft, 2025-01-23:
@hawkeyexp  if you want to update this to work with NetworkManager. That would help, otherwise I might only ship it for wpa supplicant.

b-morgan, 2025-01-24:
I downloaded 2025-01-06_2024-11-19-octopi-bookworm-arm64-lite-1.1.0.zip and installed it on an RPi 3B. When I login with SSH I get the "welcome" text including "This image comes without a desktop environment...". However, the ".../install-desktop" script isn't there. Obviously, there are two potential fixes... fix the "welcome" text or add a working script.

guysoft, 2025-01-26:
@b-morgan The script path has moved to:
[/opt/octopi/scripts/install-desktop](https://github.com/guysoft/OctoPi/blob/252e4f10b31a4251687213b1b4770a5e2c5845aa/src/modules/octopi/filesystem/root/opt/octopi/scripts/install-desktop#L4)

Will update the message, not sure I'll make an RC3 for that. will consider, perhaps for that and removing enable_network_monitor

thinkjk, 2025-02-04:
@foosel I'm testing with an rpi5 and a pi cam module 3 via MIPI and I don't know how to get the camera working. Are there any logs that I can provide you to help troubleshoot?

thinkjk, 2025-02-08:
@guysoft I've been trying everything I can think of, but can't get the camera working in octoprint.

guysoft, 2025-02-09:
@thinkjk Perhaps this change needs to be reverted for v3
https://github.com/guysoft/OctoPi/issues/837#issuecomment-2536532941

tl;dr 
I switch in ``/boot/firmware/config.txt``
from ``camera_auto_detect=1`` to ``camera_auto_detect=0``

So you can try switching in ``/boot/firmware/config.txt``
back to ``camera_auto_detect=1`` and see if that helps?

cp2004, 2025-02-09:
This is the switch between the new (current) libcamera-based webcam system and the legacy webcam system. Mjpg streamer only works with the legacy system.

For a V3 camera you need the new libcamera system, so it won't work if you have legacy enabled. Unless I've missed a change, OctoPi doesn't have support for libcamera.

When you see supported=1 detected=1 you have legacy stack. If you see libcamera interfaces=1, you're using new stack.

thinkjk, 2025-02-09:
@guysoft @cp2004 Thanks for the responses. The OctoPi Up-to image doesn't support the Rpi5. Did you mean to use the scripts in that image to update your image?

https://github.com/guysoft/OctoPi/issues/802 I've been unable to get this to work and the script creator said it's been discontinued due to the new camera stack. 

Would it be possible to get the updated camera stack added in RC3?

Thank you!

guysoft, 2025-02-25:
@thinkjk
*  #802 is still an open issue, I am not sure how to fix it, let alone for RC3.
* OctoPi Up-to image runs on the OctoPi images and updates it, not the other way round.
* [raspi-cam-srv](https://github.com/signag/raspi-cam-srv/) looks cool! I have no time to evalute it if its any good and worth shipping, if you could ask in the #802 issue perhaps people could test it out and report if its a good replacement.

foosel, 2025-02-25:
@guysoft please don't do any experiments with the camera server, especially not on a third RC. The code should be frozen apart from regression fixes at this point. 

Based on my experience with alternatives, the risk that stuff breaks for *everyone* one way or another is extremely high. Replacing the current camera stack with something different requires a long test phase related to hardware, performance and stability. And issues with the latter is also the reason why I haven't yet sent you a PR to replace the current one with the new camera stack I pushed out as an alternative OctoPi-UpToDate based image the past two years now. Modernizing the stack is frankly a project in itself, not something to quickly fix, especially since the alternative software stacks out there aren't backwards compatible, performant or stable enough yet.

Once you have a stable version of OctoPi 1.1.0 out that will be the base image I'll build OctoPi-UpToDate against, and that will also mean a new camera stack build so people can use that then with RPi Cam 3 (with the downside of it being less stable - which again is the reason it has not become the default, I'm blocked on a big issue on `camera-streamer` and the maintainer has sadly basically gone AWOL).

foosel, 2025-02-25:
After a quick look at that server you linked, it doesn't look like it has support for usb cameras but only RPI cameras, so it's a no go. Additionally, I have some doubts about the performance and that it only supports mjpeg is also not that great. Given that it's 2025 it would be preferable to also have a replacement support webrtc.

lpla, 2025-02-25:
To avoid getting it lost into the void, a year ago I suggested ustreamer as a possible replacement (quoting Debian 12 Bookworm, Raspberry Pi 5 and Camera Module 3 support from them) also because it is an active project (a release yesterday) with many contributors, support for WebRTC (h.264), direct replacement interface for mjpg-streamer as denoted in the README.md and compatible with USB cameras: https://github.com/guysoft/OctoPi/issues/823#issuecomment-2076887191

cp2004, 2025-02-25:
Ustreamer didn't have Pi Cam V3 (libcamera) support for a long time, especially when we first looked at new camera stacks. It could be useful to look at that again - worth opening its own issue somewhere.

lpla, 2025-02-25:
The issue I linked in that comment was closed by confirming that it works using the command line wrapper `libcamerify ustreamer ...`

foosel, 2025-02-25:
I might be wrong, but I'm pretty sure I tested libcamerify a while ago and the performance impact was quite bad. And libcamera itself also changed it's API a ton causing breaking in things compiled against it on updates of it from apt. But yes, wrong issue to discuss this IMHO.

guysoft, 2025-03-11:
Use RC3:
https://github.com/guysoft/OctoPi/issues/847

closed 2025-03-11 by guysoft