guysoft/OctoPi #843: RPI Imager repo links for nightlies don't distinguish bit versions

Taomyn, 2025-01-08:
Was hoping to try this some time this week, so I tried to use the RPi imager to grab the latest, but it's a bit confusing as to which 64bit one is which as some are duplicated:

[image]

I'm using the Windows imager with the command: `"C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe" --repo https://octopi.gnethomelinux.com/rpi-imager/rpi-imager-octopi.json` the same happens with the mirror link.

_Originally posted by @Taomyn in https://github.com/guysoft/OctoPi/issues/837#issuecomment-2549554021_

guysoft, 2025-01-16:
Thanks for pointing out will look in to it.

The issue is because the ending was added in the variant here:
https://github.com/guysoft/OctoPi/blob/devel/src/variants/rpios_arm64/config#L8

But now OctoPi uses CustomPiOS v2 which uses the base board system and not a variant to build base64:

https://github.com/guysoft/CustomPiOS/blob/devel/src/images.yml#L7

Need to think if there is an elegant way to add this to the broad configuration

guysoft, 2025-01-16:
Ok according to the internal sourcing diagram I should be able to just inject the architecture in to the distro name because its already sourced, will test that:

https://github.com/guysoft/CustomPiOS/wiki/CustomPiOS-internals#execution-scripts-order

guysoft, 2025-02-05:
Didn't work, I need to investigate further

guysoft, 2025-05-05:
Fixed

closed 2025-05-05 by guysoft