guysoft/OctoPi #845: 502 for raspbian.org

neunkasulle, 2025-02-17:
Seems like it has gone dark between the 7th of February and the 11.
Based of internet archive:

https://web.archive.org/web/20250211015411/https://www.raspbian.org/

Would Raspberry Pi OS be an alternative to link to instead? 
I wanted to build octopi os from scratch anyhow, so I can report around tomorrow if Raspberry Pi OS would work out of the box

guysoft, 2025-02-23:
Hey, are you talking about updating the link in the README?

> This repository contains the source script to generate the distribution out of an existing [Raspbian](http://www.raspbian.org/) distro image.

Or about something else that this effects?

b-morgan, 2025-02-23:
I think the issue is that "www.raspian.org" is returning a "502 - Bad Gateway" so anywhere you have that link it needs to be changed and the suggestion is "www.raspberrypi.com/software/". I see two such links in the README.

guysoft, 2025-02-23:
Where have you seen that link except the readme? (Where I already fixed it)

b-morgan, 2025-02-23:
> (Where I already fixed it, once)

In README.rst: Features, second bullet and Requirements, third bullet.

```
Features
--------

* `OctoPrint <http://octoprint.org>`_ host software for 3d printers out of the box
* `Raspbian <http://www.raspbian.org/>`_ tweaked for maximum performance for printing out of the box
* `mjpg-streamer with RaspiCam support <https://github.com/jacksonliam/mjpg-streamer>`_ for live viewing of prints and timelapse video creation.

Developing
----------

Requirements
~~~~~~~~~~~~

#. `qemu-arm-static <http://packages.debian.org/sid/qemu-user-static>`_
#. `CustomPiOS <https://github.com/guysoft/CustomPiOS>`_
#. Downloaded `Raspbian <http://www.raspbian.org/>`_ image.
#. root privileges for chroot
#. Bash
#. git
#. sudo (the script itself calls it, running as root without sudo won't work)
#. jq (part of CustomPiOS dependencies)
```

b-morgan, 2025-02-23:
While you are there, should the "http://" links be "https"//"?

commit 2f27bdfb (2025-02-23): Fix links to https and raspian links fixes #845
closed 2025-02-23 by guysoft