guysoft/OctoPi #847: OctoPi 1.1.0 RC3 Status

guysoft, 2025-03-11:
Third release candidate for OctoPi 1.1.0

There are both 32bit and 64bit images available for Raspberry Pi

There is also a new beta release for [Le Potato AML-S905X-CC](https://libre.computer/products/aml-s905x-cc/).
Its beta because I is not as tested as OctoPi for Raspberrypi. And I hope that having it available might let people play with it and tell me what can be done to get the wifi and cameras working correctly. 

This image brings support to Raspberry Pi 5. It also drops wpa-supplicant support due to https://github.com/raspberrypi/bookworm-feedback/issues/72
Please try the release candidate so we know it works.

Also a request for help in OctoPi - Our second mirror that was supplied by @Wille on discord has been removed, so if anyone wants to help host a second mirror for https://unofficialpi.org/Distros/ that would help a lot so we don't loose OctoPi images.

32bit armf:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly/2025-03-10_2024-11-19-octopi-bookworm-armhf-lite-1.1.0.zip

Md5: [``00613a6a64130c4aae9d2375485e3dc2``](https://unofficialpi.org/Distros/OctoPi/nightly/2025-03-10_2024-11-19-octopi-bookworm-armhf-lite-1.1.0.zip.md5).

64bit arm64/aarch64:
Download it at:
https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2025-03-10_2024-11-19-octopi-bookworm-arm64-lite-1.1.0.zip

Md5: [``74cfd8e6c5b6ff9d8443aaa357201bcd``](https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2025-03-10_2024-11-19-octopi-bookworm-arm64-lite-1.1.0.zip.md5).

Let Potato AML-S905X-CC (based on Debian):
https://unofficialpi.org/Distros/OctoPi/debian_lepotato/nightly/2025-03-10_debian-12-base-arm64+aml-s905x-cc-1.1.0.zip
Md5: [``8a6db7245e6bced1222852072dc86458``](https://unofficialpi.org/Distros/OctoPi/debian_lepotato/nightly/2025-03-10_debian-12-base-arm64+aml-s905x-cc-1.1.0.zip.md5)

### Changes in the image

* Raspberry Pi 5 support https://github.com/guysoft/OctoPi/issues/823
* Beta support for Le Potato AML-S905X-CC
* Images uses Debian bookworm
* Due to bookworkm ``/boot/config.txt`` has moved to ``/boot/firmware/config.txt`` See https://github.com/guysoft/CustomPiOS/issues/220
* Support changing default user name on rpi-imager
    * Move octoprint python virtualenv to /opt/custompios/oprint and add a symlink https://github.com/guysoft/OctoPi/commit/e62bada2c4e91372ace724f7839ab61e34bfb824
* Dropped support for wpa supplicant due to https://github.com/raspberrypi/bookworm-feedback/issues/72
Please try the release candidate so we know it works.
* New NetworkManager headless support added, its pretty basic for now since it is expected most people configure the wifi via rpi-imager. There is a new ``wifi.nmconnection`` file in the ``/boot/firmware`` folder that lets you configure wifi headless. Its more limited at the moment than the wpa supplicant version
* Remove camera ID 046d:0825 Fixes https://github.com/guysoft/OctoPi/issues/759

Changes from RC1:
* Fixed issue with webcam not working https://github.com/guysoft/OctoPi/issues/837#issuecomment-2536532941
* Fixed wifi blocking https://github.com/guysoft/CustomPiOS/commit/e553cad7e09410bf27a2083211d6e606d76efe21
* Network check has been fixed #840 (Thanks @david-forster10)

Changes from RC2:
* Updated install-desktop https://github.com/guysoft/OctoPi/commit/430a00eba263cc31c92638b9a2676509956fb366
* Remove network monitoring (networkcheck) which only works on wpa-supplicant https://github.com/guysoft/OctoPi/commit/ea5d502bff7936ffabbf8f1cab54ab9fd26ab929
* [switch to modern OS dependencies for latest numpy versions](https://github.com/guysoft/OctoPi/commit/bf506863e3b8d7c5fc7b7e1026aa25c5dd33931a)

guysoft, 2025-03-23:
No complaints so far, I think we finally have a release.

Deskwared, 2025-04-05:
It is likely that I am just missing something simple, but I cannot get this build to successfully connect via wifi or ethernet cable.  I am not a programmer, so very likely just doing something wrong.  I simply downloaded the 64 bit .img file and selected it when running the Pi imager, saving directing to a USB formatted FAT32.  I ensured that the OS customization settings were all accurate for my network, and I can see the information properly included in the firstrun SH file as well.  The ethernet lights do not even illuminate at all.  Is there a step I am missing?  (My Pi 4Bs all connect as expected, using the released OctoPi build and Pi imager).  Please don't waste much time trying to help me, if it is not a simple answer.  I was just going to try playing with the build this weekend since I had a few hours.  Thank you!

guysoft, 2025-04-06:
@Deskwared The Ethernet setting should be default if you don't enable any customization settings. I don't know how that would fail

Deskwared, 2025-04-06:
> [@Deskwared](https://github.com/Deskwared) The Ethernet setting should be default if you don't enable any customization settings. I don't know how that would fail

Huh…maybe it’s something strange in my router settings.  Beats me.  Thanks for the reply!

foosel, 2025-04-07:
Still quite busy with OctoPrint 1.11.0rcs, thus only now gotten around to throw that against the testrig and e2e tests:

- [32bit run](https://github.com/OctoPrint/devtools/actions/runs/14304020761)
- [64bit run](https://github.com/OctoPrint/devtools/actions/runs/14304436086)

All green, manual smoke test on the 64bit image also looked good (plugin installation, connection to virtual printer, update to 1.11.0rc7).

Also ran the 32bit image through the UpToDate scripts, alas, that seems to have an issue, but that's something I need to look into on my end. Just in case that's interesting: The modified image currently boots, but then immediatly shuts down again. Doesn't even seem to run firstrun script, I'm suspecting the firmware/bootloader updates, but unsure at the moment. Very hard to debug given it just shuts down, no way to access dmesg and such. That's going to be fun to debug (not), will probably need to connect a serial terminal as the screen just goes black right after.

guysoft, 2025-04-07:
@foosel do you have an image I can test? can try and find time to have a look

foosel, 2025-04-07:
Already figured out what's causing the issue, now I just need to figure out why...

[image]

It doesn't like the kernel command line that fires off the firstrun.sh script, or at least that's my current guess. Will experiment a bit. Have it on a KVM now so I can record the boot and thus see the errors.

Built image is here: https://github.com/OctoPrint/OctoPi-UpToDate/releases/tag/1.1.0rc3-1.10.3-20250407090131

But again, don't worry about that, that's on my end.

guysoft, 2025-05-05:
Released
https://github.com/guysoft/OctoPi/releases/tag/1.1.0

ammgws, 2025-05-06:
Is the "new camera stack" now the default? Or is a separate image still required for that?

[image]

guysoft, 2025-05-06:
@ammgws
Its not the default because it will break stuff for people, see this commend in RC2:
https://github.com/guysoft/OctoPi/issues/842#issuecomment-2682171723

foosel, 2025-05-06:
For the record, the reason why there's no 1.1.0 build with the new camera stack available yet is simply that `camera-streamer` currently refuses to compile on bookworm. WHich means it can't be packaged for the new image. I've been trying to find out why, but OctoPrint 1.11.0 took priority, and currently I'm on a needed emergency break.

ammgws, 2025-05-09:
@foosel Thank you for the explanation! After struggling with the mjpg-streamer and trying to get a full frame (non-cropped) image from my camera I think I will just wait for the camera-streamer enabled build to be ready. Just for reference I was able to build and run it with the PRs shown below:

```
> git clone https://github.com/ayufan/camera-streamer --recursive
> cd camera-streamer
> git checkout main
> git fetch --all
> git fetch origin pull/168/head:pr-168
> git fetch origin pull/169/head:pr-169
> git merge pr-168
> git merge pr-169
```