guysoft/OctoPi #848: CM5 or rPI 5 GPIOs are not supported because rpi.GPIO liib does not support it

NilsRo, 2025-03-16:
AS discussed here https://forums.raspberrypi.com/viewtopic.php?t=361218 the installed GPIO lib does not support PI5/CM5. This leads into issues with all plugins using it. 

```
2025-03-16 15:02:48,769 - octoprint.plugins.enclosure - INFO - mqtt helpers not found. mqtt functions won't work
2025-03-16 15:02:48,771 - octoprint.plugins.enclosure - INFO - Setting GPIO pin 5 as OUTPUT with initial value: 1
2025-03-16 15:02:48,772 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on log_error. Arguments:
('Cannot determine SOC peripheral base address',)
Traceback (most recent call last):
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_enclosure/__init__.py", line 1570, in configure_gpio
    GPIO.setup(pin, GPIO.OUT, initial=initial_value)
RuntimeError: Cannot determine SOC peripheral base address
2025-03-16 15:02:48,777 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on log_error. Arguments:
('You must setup() the GPIO channel first',)
Traceback (most recent call last):
  File "/opt/octopi/oprint/lib/python3.11/site-packages/octoprint_enclosure/__init__.py", line 953, in update_ui_outputs
    val = GPIO.input(pin) if not output['active_low'] else (not GPIO.input(pin))
                                                                ^^^^^^^^^^^^^^^
RuntimeError: You must setup() the GPIO channel first
2025-03-16 15:02:48,779 - octoprint.plugins.octopod - INFO - OctoPod loaded!
2025-03-16 15:02:48,810 - octoprint.plugins.prettygcode - INFO - Pretty GCode.
```

It could be perhaps solved using another library as mentioned in the forum post above. 

[octoprint-systeminfo-20250316145347.zip](https://github.com/user-attachments/files/19272359/octoprint-systeminfo-20250316145347.zip)

guysoft, 2025-03-17:
Hey, Not exactly an OctoPi issue, the error is coming from a plugin you are using called:
https://plugins.octoprint.org/plugins/enclosure/

I can make sure to ship a package that would help support this on OctoPi if its not too big, bug you need to take it up with the plugin developer to support it in the first place.

FYI @vitormhenrique who develops it

NilsRo, 2025-03-17:
It happens with every plugin running before with an older PI. I think it is because an old RPi.GPIO Library is used. An alternative could be perhaps preinstalled with OctoPI/venv. But did not get deeper into it yet.