guysoft/OctoPi #850: Unable to connect to hidden networks

Topless-Stang, 2025-03-22:
Fresh install of latest official build on Rpi 3B+.  Unable to connect to network that was pre-configured using the raspberry pi imager.  I am also unable to even see SSID of hidden network using wpa_cli, but BSSID does show.  If I install the latest 64 bit build of generic debian/bookworm I can connect to and also see the SSID of the same network.  I also have an RPi 4 setup with the latest official Octopi Build via the imager and it also functions fine.  The RPi4 current official Octopi build seems to be configured with access to Network manager, however the RPi3 build does not have network manager configured.  I have also tried editing the wpa-supplicant file with the scan_ssid=1 option to no avail.  In fact when using the raspberry pi imager software the wpa-supplicant file was generic and didn't have my settings at all.

guysoft, 2025-03-23:
What version of OctoPi?
Please try RC3 that uses now NetworkManager.

Topless-Stang, 2025-03-25:
I was just using the default image offered by using the Raspberry pi Imager.  Do i need to install RC3 manually?

guysoft, 2025-04-01:
ok great to hear, closing issue

closed 2025-04-01 by guysoft