guysoft/OctoPi #851: IP Unreachable Until Manual ARP

Artalius, 2025-04-24:
After OctoPi boots up, I couldn’t reach it via its IP address from non-ethernet devices — including OctoApp or accessing the Web UI on the same VLAN.

It seems like the IP isn’t announced on the network right away, so it's not reachable until something triggers an ARP broadcast.

**Workaround:**

I added this command to my `crontab` using `@reboot`:

```sh
@reboot arping -c 1 -w 2 -I $(ip route | awk '/default/ {print $5; exit}') $(ip addr show $(ip route | awk '/default/ {print $5; exit}') | grep -oP 'inet \K[\d.]+') >/dev/null 2>&1
```

After this, it works — devices can reach OctoPi and OctoPrint. I recognize that `arping` is not in the pre-installed apt stack.

Yes, the arping command was "grenerated" to make it so that `-I` was not always `wlan` and to get `IP`

**Request:**

Could OctoPi include a fix for this, maybe sending a simple ARP request on boot so the IP shows up on the network? Might help others with similar mixed network setups.

Thanks!

Artalius, 2025-04-24:
[VLAN_arp_sweep.zip](https://github.com/user-attachments/files/19881791/VLAN_arp_sweep.zip)
Zip of service, sh, and installer used

I could have added the crontab here too, but I was tired of seeing these sh files at this point.

guysoft, 2025-05-05:
Strange, never had that before. It does this on all networks you are on?