guysoft/OctoPi #853: WiFi Settings Not Being Set using Raspberry Pi Imager

d42ohpaz, 2025-04-27:
I am attempting to install OctoPi 1.0.0 with OctoPrint 1.11.0 (build 20250422094030) from the Raspberry Pi Imager, and the WiFi is not connecting. If I use the typical Raspberry Pi (Bookworm) or RetroPie images, the WiFi is set up correctly; so it must be something that OctoPi is doing differently.

I have attempted to set up the octopi-wpa-supplicant.txt manually, and I have attempted changing the file name to the standard wpa_supplicant.txt to no avail. I have compared the firstrun.sh between bookworm and OctoPi, and they are identical. I have also removed the octopi-wpa-supplicant.txt.

It took the [StackOverflow advice](https://stackoverflow.com/a/46747362) of adding "scan_ssid=1" and changing my country to `US` in the `wpa_supplicant.conf`. It seems that these values were not respected from the Raspberry Pi Imager. 

[image]

guysoft, 2025-05-05:
Not sure what is different here.

itymarcel, 2025-07-21:
For me, since 1.1.0 wifi setup is broken. There's no `wpa_supplicant.conf` anymore. Wifi data is put to `firstrun.sh` but not taken over by the Pi, it never creates the `/etc/wpa_supplicant/wpa_supplicant.conf` that it needs to connect to Wifi.