guysoft/OctoPi #855: OctoPi 1.1.0 64 bit wifi.nmconnection file example missing?

jneilliii, 2025-05-23:
I noticed a note about manually setting wifi connections using the file /boot/firmware/wifi.nmconnection, but there doesn't seem to exist this file, or firmware folder for that matter after initial flashing. 

Now that we're using networkmanager I wonder if it might be plausible to create a ap mode connection? I'm testing out the below commands I found from a RaspiWifi fork [here](https://github.com/ModestMannfred/RaspiWiFi) and it did work when bringing the connection up manually. Was thinking of creating a plugin that adds the ability to configure the connections, but think I will need to add nmcli to sudoers to allow it. 

```
nmcli con add type wifi ifname wlan0 mode ap con-name wifi-hotspot ssid OctoPi-WiFi autoconnect true
nmcli con modify wifi-hotspot connection.autoconnect-priority -999
nmcli con modify wifi-hotspot wifi.band bg
nmcli con modify wifi-hotspot wifi.channel 1
nmcli con modify wifi-hotspot ipv4.method shared ipv4.address 10.0.0.1/24
nmcli con modify wifi-hotspot ipv6.method disabled
```

jneilliii, 2025-05-23:
Screenshot of a zip download from releases, opened in 7zip to what I assume is the boot partition since it's fat. There is no firmware folder. 

[image]

jneilliii, 2025-05-25:
Let me clarify. The process I use is download zip, flash to SD card/USB thumb drive without applying settings. Once complete I only seem to get this boot partition in Windows. Is this the problem, you only see the 2nd partition you are referring to if on Linux OS?

jneilliii, 2025-05-26:
Apologies, redid a new flash and see the fie now where expected.

closed 2025-05-26 by jneilliii

FredLehto, 2025-06-02:
Hey there.

Thanks for the good software Guy. 

I landed on this thread as I was trying to find the mentioned wifi.nmconnection file. I was looking for it in the rootfs, as it is described in the documentation "on the root of the flashed card". 

Perhaps it might be clearer to mention the location (bootfs) in the documentation.

Anyway.

Thanks again for the smashing software. It is killer.

commit 7c48c2f7 (2025-06-12): Fixes #855