guysoft/OctoPi #857: /etc/systemd/system/octoprint.service needs "Environment="LG_WD=/tmp" added

b-morgan, 2025-06-19:
I am the author of [OctoPrint-FanSpeedMirror](https://github.com/b-morgan/OctoPrint-FanSpeedMirror) and while attempting to upgrade to OctoPi 1.1.0, I discovered that the scripts the plugin was calling were failing.

The journey is documented [here](https://community.octoprint.org/t/fan-speed-mirror-plugin-fails-on-octopi-1-1-0-bookworm-works-on-octopi-0-18-0-buster-and-octopi-1-0-0-bullseye/63694).

The reason for this request is because anyone who uses this plugin (or others like it) to interface with hardware is likely to run into similar issues. The hardware I'm using, [Adafruit DC and Stepper Motor HAT for Raspberry Pi](https://learn.adafruit.com/adafruit-dc-and-stepper-motor-hat-for-raspberry-pi/installing-software) requires the Python libraries that interface with the hardware be added to a Python venv and I suspect that even hardware connected to the GPIO pins of the Raspberry Pi may have a similar issue.

The final issue that @jneilliii and I encountered was `xCreatePipe: Can\'t set permissions (436) for //.lgd-nfy0, No such file or directory`. We found [this issue](https://github.com/gpiozero/gpiozero/discussions/1153#discussioncomment-9726813) which led us to the solution of adding "Environment="LG_WD=/tmp" to `/etc/systemd/system/octoprint.service`. 

I believe this change will not have a negative impact on the general OctoPi / OctoPrint population but will prevent the error for anyone using plugins that have subprocess calls in them. Another plugin would be [OctoPrint-GCodeSystemCommands](https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands).

guysoft, 2025-06-20:
Sounds reasonable. Do you want to draft a PR?

jneilliii, 2025-06-20:
PR submitted.